stanleyhuangyc/ArduinoOBD #23: OBD-II UART Adapter - `odb.init()` keep failing

Jossef, 2015-03-13:
I recently bought [this](http://freematics.com/pages/products/arduino-obd-adapter/) OBD-II arduino adapter.

### I'm troubleshooting with the [rpm-led basic example](https://github.com/stanleyhuangyc/ArduinoOBD/blob/master/libraries/OBD/examples/rpm_led_uart/rpm_led_uart.ino)

iv'e defined `#define DEBUG Serial` **and saw these messages** in arduino ide serial monitor:

```
SEARCHING...UNABLE TO CONNECT>[865698]ATZ OBDUART v1.0>[865754]ATE0 
OBDUART v1.0>[865754]ATE0 
ATEOOK)[865807]ATL1
OK>[865860]010C
SEARCHING...UNABLE TO CONNECT>[872586]ATZ OBDUART v1.0>[872641]ATE0 
OBDUART v1.0>[872641]ATE0 
ATEOOK)(872694]ATL1
OK)(872748]010C
SEARCHING...UNABLE TO CONNECT>[879473]ATZ OBDUART v1.0>[879528]ATE0 
OBDUART v1.0>[879528]ATE0 
ATEOOK)(879582]ATL1
OK)(879635]010C
SEARCHING...UNABLE TO CONNECT>[886362]ATZ OBDUART v1.0>[886417]ATE0 
```

### this is how i connected the OBD-II adapter 
tested on Arduino Leonardo, 
My car - *Volkswagen Polo 1998*

[image]

stanleyhuangyc, 2015-04-20:
1998 Polo is not supported

conors891, 2015-08-04:
is 2000 mk2 punto supported?