stanleyhuangyc/ArduinoOBD #35: Maintain this issue list please?

rubytastic, 2015-12-27:
Looked in buying several hardware units for a personal project.
This issue tracker not updated after a year does not make me feel good as a new customer.

Could someone please clean it up and update the issues and pull requests?

M-tech-Creations, 2015-12-28:
Judging from how long it took there customer service to react / respond and what I needed to do in order to hear back from them, this request might take a while to be fulfilled. If it happens.

rubytastic, 2015-12-28:
@M-tech-Creations 
Shame this solution seems the only open source odb solution out there.
im in need of 50 of these units but will not buy them if this repo isnen't even maintained ( months of backlog and pull requests )