stanleyhuangyc/ArduinoOBD #37: OBD-II UART does not work

Mondeo, 2016-02-29:
Hi,
I received the OBD-II UART Adapter designed for Arduino.
I loaded up the "rpm_led_uart.ino" sketch to my Arduino Uno. The sketch should switch Pin 13 to high if the engine turns faster then 3000rpm.
I tested today but nothing is happen. On the Adapter a red LED is on and on the Arduino the green LED "On" shines. The Arduino is wired as follows: red wire to +5 Volt,  black wire to GND, white wire to Rx (Pin 0) and green wire to Tx (Pin 1). My Car is a Mitsubishi Colt 1.3 16V. 
Can you explain me what is wrong and how can I check the installation.
Thank you and best regards. Urs

kg6hxm, 2017-01-17:
Are you able to see the raw serial data?
Also, the TX/RX lights should flicker. Do they?

Lacassefire, 2017-03-24:
Hi, 
I got OBD2UART V2 connector, plug it to arnuino uno, install the library OBD2UART. Load the code for red light. Test it on Honda civic 2012 and on BMW 330ci 2001. And it's does't work and the result is the same.

-When OBD2 adapter is plug in the car the red led is on.
-On the board I see power signal.
-I don't get Tx/Rx signal light did not flicker.
-Check led pin 13 flash once in the begining.

Did you what should I try to make it work?
How did you get the raw data?

Thank you

superlazycoder, 2017-05-30:
I am having this same issue. I have uncommented #define DEBUG Serial in OBD2UART.cpp and the output is as follows.

`<<<ATI

>>>?C��
<<<ATI

>>>O(��MA*��.j
�
<<<ATI

>>>O(��MA*��.j
�
<<<ATI

>>>
<<<ATI

>>>
`

firepower2k11, 2018-01-21:
The output looks like you use the wrong baudrate.
If you're using the Arduino IDE, down right on the serial output window change the baudrate to 11520 (should be the correct one) or 9200.

aaronchih, 2018-06-13:
Hello.
I have the same issue, did you solve this?
[image]if you have solution, help me