stanleyhuangyc/ArduinoOBD #43: File does not contain valid library

sterlija, 2017-05-12:
I downloaded the zip file. My computer automatically uncompressed it. I removed the "master" and then tried to go into arduino -->add library--> zip file and it just gives me "Specified folder/zip file does not contain a valid library". Does anyone know how to fix this?

superlazycoder, 2017-05-30:
The contents of the library folder within the git repo, just needs to be placed into {Documents}/Arduino/libraries

After the folders are in there, "such as the OBD folder" then you will be able to #include <OBD.h> and use the libraries as if they were part of your sketch

(I am on a mac, this may be different for windows, but I doubt it is that much different)

sterlija, 2017-05-30:
Thank you,

I am actually on a Mac as well. So hopefully I can get it to work. Probably just me. 

Thank you,
Jacob Sterling