stanleyhuangyc/ArduinoOBD #50: OBD-II UART Adapter V2 - Data reading issue

laremy, 2018-02-17:
Hi everyone, 

When I tried to upload your example sketch (for reading PID's) with the OBD connected to my Arduino UNO, it did not work, so I uploaded the sketch without connected the OBD and after that I connected the OBD to my Arduino UNO and obtained a succession of : **ARTZAR..................................................**

Does anyone know what is happening?
Many thanks

firepower2k11, 2018-02-17:
Please change the baud rate in the Arduino IDE from 9200 to 115200 and try again.

laremy, 2018-02-17:
It was already 115200 baud rate :( Still have this kind of thing : 
ATZ
ATZ
ATE0
ATH0
010D
ATR
ATZ
ATZ
ATE0
ATH0
010D
ATR
ATZ
ATZ
ATE0
ATH0
010D
ATR
and it does not stop...

I see the blue light speaking with my Arduino UNO but I do not obtain the PID's values

I do not know where is the problem

laremy, 2018-02-17:
This is what I did :

1) compiled your sketch
2) then connected the OBD to my Arduino UNO : white -> Tx Arduino UNO, Green -> Rx Arduino UNO, Black -> GND and Red -> 5V
3) then I open the serial monitor (115200 baud)

And I obtain this : 
ATZ
ATZ
ATE0
ATH0
010D
ATR
ATZ
ATZ
ATE0
ATH0
010D
ATR
ATZ
ATZ
ATE0
ATH0
010D
ATR

Do you have an idea?

Many thanks

laremy, 2018-02-17:
Still trying but still have the same result :( Hope that you will help me.

Kind regards

laremy, 2018-02-18:
and when I tried to upload the sketch with the OBD connected to the Arduino UNO, I obtained : 

avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x41
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x54
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x49
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x0d
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x41
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x54
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x49
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x0d
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x41
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x54
An error occurred while uploading the sketch

But before I had 

avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x00
An error occurred while uploading the sketch

Is it possible that I did a wrong manipulation and corrupted the OBD ?

Does I have to other anotther OBD2 Uart Adapter?

Kind regards

laremy, 2018-02-18:
Hi everyone,

I have an OBD2 Uart adapter commanded on Freematics. My problem is the following (I begin with Arduino)

1) First I connected the adapter to my Arduino UNO (Rx/Tx) and when uploading I had that : 

avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x00
An error occurred while uploading the sketch

2) Then I did some bad manipulation instead of connecting the adapter to Rx/Tx, I connected to A2 A3 and uploaded the same sketch => nothing happened

3) Then retried with Rx/Tx connection and had that (no more 0x00) :

avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x41
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x54
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x49
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x0d
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x41
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x54
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x49
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x0d
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x41
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x54
An error occurred while uploading the sketch

Do you think that I corrupted the Adapter with step 2?

I know now that Rx/Tx do not have to be connected while uploading, so I did it and then reconnected the Adapter to Rx/Tx and instead of having in the serial monitor :

"the right text....." I have a succession of 
ATZ
ATZ
ATE0
ATH0
010D
ATR
ATZ
ATZ
ATE0
ATH0
010D
ATR
ATZ
ATZ
ATE0
ATH0
010D
ATR

I am using com3 with the good baud rate (115200 in my case)

I do not understand what is it happening?

1) Did I corrupt the adapter with my wrong manipulation (the light of the adapter still works when I connect it)
2) Is it the sketch (right compilation but problem with the reading in the serial monitor)
3) Any other idea?

If you need more detail,
Feel free to ask me

Kind regards

firepower2k11, 2018-02-18:
As far as I can see it's working correctly - the ATZ (...) are the AT commands to send messages to the car.
If you visit the description of the adapter again there is the list of supported commands linked:
https://www.sparkfun.com/datasheets/Widgets/ELM327_AT_Commands.pdf

Which protocol does your car use btw.?

laremy, 2018-02-18:
Many thanks for you answer

The car use the European protocol, ISO9141-2

I tried with and without connecting the OBD to the car and the results are the same.

I did the following steps : 
1) uploaded your sketch on my Arduino Uno
2) then connected the Obd to my Arduino (black => ground, red => 5v, white => Arduino Uno Tx and Green => Arduino Rx)
3) then open the serial monitor with 115200 baud rate selection

Is it the right way to read the sketch?

As I precised, I did a wrong manipulation by connecting the white and the green on A2 and A3 and the syn message changed (from not in sync: resp=0x00 to not in sync: resp=0x41...). I did it when I though that the obd had to be connected to the Arduino Uno before uploading the sketch. But it is wrong to do that.

Did I break the adapter or deprogram it? The light on the Adapter still works and seems to speak with my Arduino Uno.

Is it a good idea to test the adapter on an Aduino mega?

Many thanks for your return

Laurent

laremy, 2018-02-18:
Moreover as you mentioned, the obd sends information to my Arduino Uno : 

ATI
ATI
ATZ
ATI
ATH0
ATRV
0100
010C
0902
ATTEMP
ATZ
ATE0
ATH0
010D
ATR
ATZ
ATZ
ATE0
ATH0
010D

So I guess that the obd still works

firepower2k11, 2018-02-18:
You say it's "ISO9141-2"... is your car using K-Line to communicate?
If yes your adapter will never work since the builtin transceiver doesn't support K-Line.

laremy, 2018-02-18:
Hi,

Many thanks for this information, I understand better.

Yes, it uses K-line (Pin 7 - ISO 9141-2 K Line and Pin 15 - ISO 9141-2 L Line = European protocol) but it is the ISO9141-2 protocol (European one) and in the datasheet of the OBD2 Uart adapter V2.1 you say that the adapter supports the ISO9141-2 protocol, is it true? So I do not understand the question about the k-line. 

But normally (even if the adapter does not support ISO9141-2 protocol), when connecting the Adapter to the Arduino Uno without connecting to te car, I will have receive text like "Freematics obd2 adapter detected" instead of a succession of AT commands? Is it true?

If you want I can send your a video

Thank you for your return and the time you spend for me I really appreciate, hope it will work :)

Laurent

aaronchih, 2018-06-13:
Hello.
I have the same issue.
[image]
How can i solve the issue.

firepower2k11, 2018-07-01:
Hi Laurent,

the code may speak the ISO9141-2 PROTOCOL but your car has K-line as PHYSICAL connection.
So they are two completely different things. The adapter itself only supports OBD-II as physical connection.
I was wondering too why I got no connection and after taking the adapter apart and searching for a datasheet of the built in tranceiver I found out, that it only connects to OBD, not K-line.

If you read the example code there is no line where it would return something like "Freematics Adapter connected", it only returns the commands which it sends and what data it receives, as far as I can see.

ALEEF02, 2020-08-05:
@firepower2k11 I am aware that this is an old thread, but I'm having a similar issue to this, and I am getting the message `Freematics OBD-II Adapter detected`. Perhaps you can help solve the issue from here... https://github.com/stanleyhuangyc/ArduinoOBD/issues/70