stanleyhuangyc/ArduinoOBD #70: Freematics OBD-II UART Adapter V2.1 CANNOT Initialize!

ALEEF02, 2020-08-05:
Hi, I spent over a month waiting for this item to arrive at my doorstep due to constant package delays. I am now extremely frustrated because I cannot get the OBD reader to initialize. I have double-checked all the connections and am trying to debug the reader and why `obd.init()` keeps failing. I am using a _slightly_ modified version of the [obd_uart_test](https://github.com/stanleyhuangyc/ArduinoOBD/blob/master/libraries/OBD2UART/examples/obd_uart_test/obd_uart_test.ino), which just replaces the Serial outputs with screen output. Everything appears fine until the commands `0100`, `010C`, and `0902` are sent. Those return `UNABLE TO CONNECT`. Mems afterwords still returns `Yes`. From there on, I am stuck in a loop, waiting for the reader to initialize. I'm kinda lost here, so any and all help would be welcome. Thanks!
[image]
```
  tft.println("Connecting...");
  
  for (;;) {
    delay(10000);
    byte version = obd.begin();
    tft.print("Freematics OBD-II Adapter ");
    if (version > 0) {
      tft.println("detected");
      tft.print("OBD firmware version ");
      tft.print(version / 10);
      tft.print('.');
      tft.println(version % 10);
      break;
    } else {
      tft.println("not detected");
    }
  }

  testATcommands();
  hasMEMS = obd.memsInit();
  tft.print("MEMS:");
  tft.println(hasMEMS ? "Yes" : "No");

  do {
    tft.println("Initializing...");
  } while (!obd.init());
  tft.println("OBD connected!");
```

mitchschaft, 2020-08-05:
It's because the code hasn't been updated in years. I've never gotten it to work and just use Tunerview.

ALEEF02, 2020-08-05:
@mitchschaft Thank you for the timely reply! Could you link me to that library?

firepower2k11, 2020-08-30:
@ALEEF02 According to the displayed messages your adapter itself works so far but it can't communicate with the car.
Which protocol does your car use?
If you're using ISO9141-2 over K-Line it will never work, as I already wrote in the other thread you mentioned me.

ALEEF02, 2020-08-31:
@firepower2k11 My car uses the SAE J1850 VPW protocol, which I learned in an email thread with @stanleyhuangyc. Since then he has stopped replying to said emails. I was at first wondering if I could get a refund, at which he stopped replying. Now I am wondering if there is a hardware or software modification that can be made to the adapter so that it can use the SAE J1850 VPW protocol. I tried modifying the library by adding PROTO_SAE_VPW = 2 to the OBD_PROTOCOLS list, but that did not help. If you have any ideas for this, that would be wonderful. Thanks :D