stanleyhuangyc/ArduinoOBD #71: OBD 1 ?

marianpolek, 2020-08-14:
Hi All, 

i have an freemantics uart adapters v2.1 OBD2 .... and mazda mx5 nb 1998. 
So obd is version 1 , in engine bay place. I have reduction from OBD1 to OBD2 . 
And i tried to run example code which light led when RPM is higher then 3000 . 
But led light never appear. 

freemantics blue led is turned on, so this is great. Arduino is ready and program is loaded to arduino. 

Do you have any idea where can be problem ? 

Thank you

marianpolek, 2020-08-14:
i found that obd.init never happen succesfully ... but i dont know why.

Lacassefire, 2020-08-15:
I get the same result on e46...

NickDeBeenSAE, 2024-02-06:
Hello, please see my name above.

I am requesting code that can read an OBDI ports code and extract it.